home-assistant/operating-system #4234: [nvme crashes on haos higher than v13.2] raspi5

gkathan, 2025-08-15:
### Describe the issue you are experiencing

i run various raspi5 with nvme hats which work perfect with HAOS 13.2 but everything higher than that (14, 15, 16) have the same issue after some days => crashing nvme ssd with HA still somehow working but eg no access to any addon (whitescreen) or on trying to edit a file with file ditor it says " ...file not existing..." and looking on the terminal it lists continouos ssd/nvme errors (see screenshot) 

* after reboot it might work again some hours / days but then fails again
* when downgrading to HAOS 13.2 everything is rock stable again

also had a dialog with chatgpt which suspects that it has to do something with latest kernel updates within HAOS which could cause this behaviour

=> there are several reddit channels / other threads on the web describing this issue and i am wondering whether you are aware of that and thinking about resolving this issue => happy to contribute more details as i can somehow reproduce the issue

cheerz from vienna

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

16.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

* raspberry pi 5 8gb
* POE M.2 HAT+ (B) Kombiniert PCIe zu M.2 und Power Over Ethernet POE für Raspberry Pi 5, Unterstützt NVMe M.2 2230/2242/2260/2280 SSD, IEEE 802.3af/at POE Netzwerk Standard
* Crucial T500 SSD 1TB PCIe Gen4 NVMe M.2 Interne SSD,

### Steps to reproduce the issue

1.upgrade from HAOS13.2 to any version higher (incluinf 16.1)
2.just wait some hours / days
3.system crashes

[image]

...

### Anything in the Supervisor logs that might be useful for us?

```txt
did not find any usefull stuff there - might have to look in more detail - but see my screenshot
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

wolverinevn, 2025-08-16:
Same issue here. Pi5 with SDD hat, nothing special.
After power loss, Pi5 boots fine and hass starts  but the disk might be locked at some points therefore stat integration and addons were unable to start. Cant even restart hass or Pi5 from supervisor.
A cold boot solves the issue.
It's a random behavior.

gkathan, 2025-09-08:
> does anybody have clue whether there is some sort of config tweak to get
> rid of those crashes ? maybe to give the newer and apparently more
> aggressive kernel some more defensive settings (as haos 13.2 with der
> Kernel runs rock solid with same hardware) thanx
>
> I have switched NVMe drives... bought pricier Samsung 980 250GB NVMe, and
> issue is no more, I hope.
> 7 days since last host start/reboot (did some changes regarding fan
> speed), and still running without single issue.
> 3x knock.
>
> So, maybe in the end, it is hardware issue.

i see ... so i will give your suggestion a try and we will watch both of us
how it works out
or is this raspberry nvme ssd maybe even better ?
thanx from vienna

ThePatzen, 2025-11-04:
Hi!

I hab the same issue.. Crash after same days uptime..

Not using RaspberryNvme...

rpi-eeprom-update solved it for me!

kommando828, 2025-12-26:
Booted into Debian and rpi-eeprom-update got me

BOOTLOADER: up to date
   CURRENT: Wed  5 Nov 17:37:18 UTC 2025 (1762364238)
    LATEST: Wed  5 Nov 17:37:18 UTC 2025 (1762364238)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)

So then ran

sudo apt full-upgrade

Sudo reboot

Reconnected and reran 
rpi-eeprom-update

*** UPDATE AVAILABLE ***

Run "sudo rpi-eeprom-update -a" to install this update now.

To configure the bootloader update policy run "sudo raspi-config"

BOOTLOADER: update available
   CURRENT: Wed  5 Nov 17:37:18 UTC 2025 (1762364238)
    LATEST: Mon  8 Dec 19:29:54 UTC 2025 (1765222194)

Then ran sudo rpi-eeprom-update -a

*** PREPARING EEPROM UPDATES ***

BOOTLOADER: update available
   CURRENT: Wed  5 Nov 17:37:18 UTC 2025 (1762364238)
    LATEST: Mon  8 Dec 19:29:54 UTC 2025 (1765222194)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.
   CURRENT: Wed  5 Nov 17:37:18 UTC 2025 (1762364238)
    UPDATE: Mon  8 Dec 19:29:54 UTC 2025 (1765222194)
    BOOTFS: /boot/firmware
'/tmp/tmp.xfPninaVK3' -> '/boot/firmware/pieeprom.upd'

Will reinstall SSD and see how long before issues appear.

kommando828, 2025-12-27:
24 hrs and no issues especially with the sqlite database which was the point at which access to files was a problem.

indigomejor, 2025-12-28:
> No luck yet with all the suggestions.
> 
> I am unsung a crucial ssd and the argon one v5...

[image] Samsung 980, Argon 5 Neo. its about the SSD, Crucial is not OK apparently