home-assistant/operating-system #4288: ZHA + Conbee 3 not working after 16.0

vilmar-hillow, 2025-09-09:
### Describe the issue you are experiencing

After updating to 16.1 initially a while ago, ZHA failed to startup with message `Failed setup, will retry: Failed to deliver packet: <TXStatus.APS_NO_ACK: 167>`. Since this was directly after OS update and no other changes, I stumbled across #4228 , thought it was related and rolled back to 16.0 which brought things back to normal. After 16.2 was released, I tried updating again only to see same error messages.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

16.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Rpi 5 + Conbee 3 stick.

### Steps to reproduce the issue

1. Update from 16.0 to 16.1 or 16.2

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2025.9.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | aarch64
user | root
virtualenv | false
python_version | 3.13.7
os_name | Linux
os_version | 6.12.34-haos-raspi
arch | aarch64
timezone | America/Toronto
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2158
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.09.0
agent_version | 1.7.2
docker_version | 28.3.3
disk_total | 916.1 GB
disk_used | 20.3 GB
nameservers | 192.168.50.1
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Cloudflared (5.3.8), Grafana (11.0.0), Samba share (12.5.2), InfluxDB (5.0.2), Studio Code Server (5.19.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 3
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth3bcc3fa (disabled), vethf6ac6b4 (disabled), vethb5b6818 (disabled), vethc3a70be (disabled), vetha9743bc (disabled), veth8bb5704 (disabled), veth852b4fb (disabled), vetha250d4d (disabled), veth39f0082 (disabled), veth256fca7 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.50.135/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth3bcc3fa (), vethf6ac6b4 (), vethb5b6818 (), vethc3a70be (), vetha9743bc (), veth8bb5704 (), veth852b4fb (), vetha250d4d (), veth39f0082 (), veth256fca7 ()
ipv6_addresses | lo (::1/128), end0 (fe80::67b8:e32:c439:38b7/64), hassio (fe80::c42a:74ff:fe91:8143/64), docker0 (fe80::507c:28ff:feac:1f64/64), veth3bcc3fa (fe80::2877:8fff:fe86:9bb9/64), vethf6ac6b4 (fe80::b4b4:fbff:fe5e:a268/64), vethb5b6818 (fe80::8495:66ff:fec7:ef5b/64), vethc3a70be (fe80::4c94:f9ff:fe2f:8726/64), vetha9743bc (fe80::b0ea:2dff:fec9:a50a/64), veth8bb5704 (fe80::5496:d6ff:fe5f:aee1/64), veth852b4fb (fe80::14da:d8ff:feab:83f8/64), vetha250d4d (fe80::6c2c:c9ff:fe57:83c3/64), veth39f0082 (fe80::1060:dff:fe5b:3dfb/64), veth256fca7 (fe80::8c0f:d5ff:fe72:339a/64)
announce_addresses | 192.168.50.135, fe80::67b8:e32:c439:38b7

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 22, 2025 at 2:49 AM
-- | --
current_recorder_run | September 8, 2025 at 11:55 PM
estimated_db_size | 342.55 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

sairon, 2025-09-09:
This seems unrelated to the OS bug you linked - in that case the integration would fail earlier as the device wouldn't appear in the system at all. Is the problem reproducible in 16.2 also after a full power cycle? There are multiple reports way from the past where this error was caused by interference, so it may be intermittent too. Preferably use the USB 2.0 (black) port on your Pi and connect the dongle with an extension cord to minimize this effect.

vilmar-hillow, 2025-09-09:
> This seems unrelated to the OS bug you linked - in that case the integration would fail earlier as the device wouldn't appear in the system at all. Is the problem reproducible in 16.2 also after a [...]

Fascinating, an extension cord worked. Weird that it wasn't an issue in 16.0 at all! Thanks for the tip =)

closed 2025-09-09 by vilmar-hillow