home-assistant/operating-system #4310: Error loading /config/configuration.yaml: [Errno 24] No file descriptors available:

denisb88, 2025-09-15:
### Describe the issue you are experiencing

Good evening, for about a week now, my connection to HA has been blocked. Nabu Casa is no longer responding, pings are working randomly, and downloading updates is impossible. If I can't restart HA, I get this message that prevents me from continuing:
`Error loading /config/configuration.yaml: [Errno 24] No file descriptors available: ‹ /config/automations.yaml ›`
I have to abruptly stop my Proxmox container and restart it.
Méthode d'installation Home Assistant OS
Core 2025.9.3
Supervisor 2025.09.0
Operating System 16.2
Interface utilisateur 20250903.5

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.12.43

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Container with Proxmox server on Ryzen mini pc

### Steps to reproduce the issue

The problem occurs randomly after a few hours of operation

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-09-15 20:42:42.686 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 20:47:42.691 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 20:52:42.691 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 20:57:42.692 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:02:42.690 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:07:42.691 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:09:46.996 INFO (MainThread) [supervisor.host.info] Updating local host information
2025-09-15 21:09:47.372 INFO (MainThread) [supervisor.host.services] Updating service information
2025-09-15 21:09:47.374 INFO (MainThread) [supervisor.host.network] Updating local network information
2025-09-15 21:09:47.415 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2025-09-15 21:09:47.417 INFO (MainThread) [supervisor.host.manager] Host information reload completed
2025-09-15 21:11:10.700 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-09-15 21:12:42.686 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:17:42.688 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:22:42.981 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:27:42.687 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:32:42.689 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:37:42.686 WARNING (MainThread) [supervisor.addons.options] Unknown option 'availability' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2025-09-15 21:39:16.881 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2025-09-15 21:39:16.882 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2025-09-15 21:39:16.884 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2025-09-15 21:39:16.884 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2025-09-15 21:39:16.884 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2025-09-15 21:39:16.884 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2025-09-15 21:39:16.935 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2025-09-15 21:39:16.936 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2025-09-15 21:39:16.936 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2025-09-15 21:39:16.936 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2025-09-15 21:39:16.936 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2025-09-15 21:39:16.937 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disk_lifetime/system
2025-09-15 21:39:16.938 INFO (MainThread) [supervisor.resolution.check] System checks complete
2025-09-15 21:39:16.938 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2025-09-15 21:39:16.981 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-09-15 21:39:16.982 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2025-09-15 21:39:16.982 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2025-09-15 21:41:16.491 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
[... 178 lines cut ...]
announce_addresses | 192.168.1.222, 2a01:e0a:971:4fa0:b547:d6f6:8bc0:3aaa, fdb1:326e:78ec:1189:8cf6:ead0:aa82:5055, fe80::b22b:8347:bb9d:9158

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 4 septembre 2025 à 13:45
-- | --
current_recorder_run | 15 septembre 2025 à 22:32
estimated_db_size | 8103.07 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 1.0.21
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 0
total_devices | 0

</details>

tomkolp, 2025-09-16:
I have the same problem, also HAOS. Do you have the cloudflared addon?

Error loading /config/configuration.yaml: [Errno 24] No file descriptors available: '/config/secrets.yaml'[image]

tomkolp, 2025-09-16:
```
Engine	Pod	Name	Status	Uptime	CPU%	MEM	/MAX	IOR/s	IOW/s	RX/s	TX/s	Command
docker	-	homeassistant	running	yesterday	15,1	1.81G	7.54G	0b	74Kb	NaNb	NaNb	/init
docker	-	addon_45df7312_zigbee2mqtt_edge	running	6 days	1,6	110M	7.54G	0b	42Kb	1Kb	1Kb	/sbin/tini -- /docker-entrypoint.sh
docker	-	addon_a0d7b954_glances	running	6 days	0,7	56.5M	7.54G	0b	0b	NaNb	NaNb	/run.sh
docker	-	addon_9074a9fa_cloudflared	running	5 days	0,7	32.2M	7.54G	0b	0b	2Kb	3Kb	/init /run.sh
docker	-	addon_core_mosquitto	running	6 days	0,4	16.4M	7.54G	0b	0b	7Kb	7Kb	/init
docker	-	addon_de91e161_hassio_onedrive_backup	running	6 days	0,2	143M	7.54G	0b	0b	0b	0b	dotnet onedrive-backup.dll
docker	-	hassio_dns	running	6 days	0,1	22.9M	7.54G	0b	0b	8Kb	37Kb	/init
docker	-	addon_core_configurator	running	6 days	0	24.5M	7.54G	0b	0b	0b	0b	/init
docker	-	hassio_audio	running	6 days	0	11.9M	7.54G	0b	0b	0b	0b	/init
docker	-	hassio_multicast	running	6 days	0	1.54M	7.54G	0b	0b	NaNb	NaNb	/init
docker	-	hassio_supervisor	running	6 days	0	1.87G	7.54G	0b	0b	0b	0b	/init
docker	-	addon_5c53de3b_esphome	running	6 days	0	419M	7.54G	0b	0b	NaNb	NaNb	/init
docker	-	hassio_observer	running	6 days	0	6.00M	7.54G	0b	0b	0b	0b	/usr/bin/observer
docker	-	hassio_cli	running	6 days	0	4.27M	7.54G	0b	0b	0b	0b	/init
docker	-	addon_core_ssh	running	5 days	
```