home-assistant/operating-system #4311: Set theme for homeassistant, but every time I open the app it goes back to default?

Daottuan, 2025-09-18:
### Describe the issue you are experiencing

Set theme for homeassistant, but every time I open the app it goes back to default?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

2025.9.3

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Set theme for homeassistant, but every time I open the app it goes back to default?

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Set theme for homeassistant, but every time I open the app it goes back to default?
```

### Anything in the Host logs that might be useful for us?

```txt
Set theme for homeassistant, but every time I open the app it goes back to default?
```

RedNo7, 2025-09-25:
my iPhone rests the theme back to default every few days - very frustrating.  I do not have a rotating IP address *=(which some people have said can be a cause of this).

agners, 2025-09-29:
This is unlikely to be OS related. Please post the issue on the respective compagnion app issue tracker (either http://github.com/home-assistant/android or https://github.com/home-assistant/ios).